kevinawalsh/logisim-evolution #17: feature request: values and buses larger than 32 bits

kevinawalsh, 2019-02-04:
This is non-trivial, because 32-bit integers are used extensively to hold Values.

TChapman500, 2019-10-23:
This would be difficult to do with Java anyways, since the numbers are always unsigned.  But this would be a welcome change.

TChapman500, 2020-09-06:
The original Logisim Evolution now has 64-bit data buses.  I might be able to cook-up a solution though.  I just created another fork and I may eventually get to.

kevinawalsh, 2020-09-06:
Which version/fork are you referring to that has 64-bit buses? Do you have a pointer?

TChapman500, 2020-09-06:
The version that you directly forked from.

BFH-ktt1, 2020-09-06:
@kevinawalsh : The 64 bit buses were introduced with push-request #327 in evolution. There were some small bugfixes afterwards.