kevinawalsh/logisim-evolution #33: Truttable automatic cursor movement on typing

BFH-ktt1, 2019-03-01:
In the current implementation the cursor will not move when typing "0", "1", and "X" int the truth table.
This can be fixed by modifying TableTabCaret.java (starting from line 262):
Remove:
if (!markA.isValid() || !markB.isValid())
				return;
			Rectangle s = getSelection();
			int row = cursor.row;
			int col = cursor.col;
			if (dy > 0) { // advance down
				col = s.x;
				if (++row >= s.y + s.height)
					row = s.y;
			} else if (dx > 0) { // advance right
				if (++col >= s.x + s.width) {
					col = s.x;
					if (++row >= s.y + s.height) {
						row = s.y;
					}
				}
			} else if (dx < 0) { // advance left
				if (--col < s.x) {
					col = s.x + s.width - 1;
					if (--row < s.y)
						row = s.y + s.height - 1;
				}
			}
			Pt oldCursor = cursor;
			cursor = new Pt(row, col);
			repaint(oldCursor, cursor, markA, markB);
			scrollTo(cursor);
    }

and replace it with:

			AutoMoveCursor(dx,dy);
		}
		
		private void AutoMoveCursor(int dx , int dy) {
			int row = cursor.row;
			int col = cursor.col;
			int InputCols = table.getInputColumnCount();
			int outputCols = table.getOutputColumnCount();
			int NrOfEntries = table.getRowCount();
			int newRow = row+dy;
			int newCol = col+dx;
			int maxcol = col < InputCols ? InputCols : InputCols+outputCols;
			int mincol = col < InputCols ? 0 : InputCols;
			if (newCol >= maxcol) {
				newCol = mincol;
				newRow++;
			}
			if (newCol < mincol) {
				newCol = maxcol-1;
				newRow--;
			}
			if (newRow < 0)
				newRow = NrOfEntries-1;
			if (newRow >= NrOfEntries)
				newRow = 0;
			Pt oldCursor = cursor;
			cursor = new Pt(newRow, newCol);
			repaint(oldCursor, cursor, markA, markB);
			scrollTo(cursor);
		}

kevinawalsh, 2019-03-07:
If you select a region of the truth table (mouse click/drag), the cursor will advance through that region as you type. And it will stay within the region you have selected. 

But you are right -- without selecting a region, it does not advance. That was intentional, but if you think it's better to advance when nothing is selected, I'm happy to consider changing it, but probably only for the case when no region is selected.

There are a few bugs in the code you posted (hitting enter now moves diagonally right/down, and there are formatting / code-style issues). I was able to just re-use the existing code almost entirely intact and getting both new and old behavior by basically just swapping out the one piece:

```
if (!markA.isValid() || !markB.isValid())
    return;
Rectangle s = getSelection();
```
For something more like this:
```
Rectangle s = (marked() && !markA.equals(markB)) ? getSelection() : getRegion();
```

Thanks for the suggestion.

closed 2019-03-07 by kevinawalsh

BFH-ktt1, 2019-03-09:
You're completely right. I posted this as in the original logisim you could click on an output entry and than fill-in with the keys (0/1/x) the complete table. In your version you need to select the region and than you have the same behavior. As when I wrote this issue I was not aware of this new "behavior" I was a bit surprised :-).
As a good alternative (besides your solution posted above) I found following (that will mark the output row below the selected cursor):

public void mouseClicked(MouseEvent e) {
			if (cursor.isValid() && cursor.col >= table.getInputColumnCount() && !marked()) {
				/* We clicked inside the output region; we mark the complete 
				 * region below the cursor.
				 * markA is already set, so we set markB to the end of the table 
				 */
				markB = new Pt(markA.row,table.getColumnCount()-1);
				repaint(markA,markB);
			}
		}

hope it makes some sense and adheres to the code-style