kevinawalsh/logisim-evolution #39: bug: Analyzer input/output tab update on file load

BFH-ktt1, 2019-03-30:
When being in the analyzer window on the input/output tab and loading a truth table file, the input/ and output window will only show the entries that were visible before. the refresh/model invalidation is not performed correctly. Switching tabs will correct the behavior (hence switching to truth table and than back to input/output will show the correct contents. I did not have the time to figure out the problem, but will post as soon as I pinpointed it out.

BFH-ktt1, 2019-04-03:
Found the problem, the VariableTab is not updated on table import. To fix this issue, the VariableTab update can be performed in Analyzer.java by:

    private void update() {
      TruthTable tt = model.getTruthTable();
      buildCircuit.setEnabled(tt.getInputColumnCount() > 0 && tt.getOutputColumnCount() > 0);
      exportTable.setEnabled(tt.getInputColumnCount() > 0 && tt.getOutputColumnCount() > 0);
      exportTex.setEnabled(
          tt.getInputColumnCount() > 0
              && tt.getOutputColumnCount() > 0
              && tt.getRowCount() <= AnalyzerTexWriter.MAX_TRUTH_TABLE_ROWS);
      ioPanel.updateTab();
    }

adding the last line in this handle of the TableListener (e.g. ioPanel.UpdateTab()).

kevinawalsh, 2019-04-15:
Fixed. Thanks for the report.

closed 2019-04-15 by kevinawalsh