kevinawalsh/logisim-evolution #41: Misunderstanding on notation in equations?

BFH-ktt1, 2019-03-30:
I noticed that in the current holycross version the Engineering notation uses  overline for not, ^ for and and v for or, whilst mathemetics uses +,.,and ¬ for not. Is this not a mix up for the not? Should it not be (¬,^, and v) and (+,., overline)?

BFH-ktt1, 2019-04-04:
Also the programming notation gives me some questions:
D:3==!A&&!B:3&&!B:2
might lead to some misinterpretation, first I think that the == is misplaced as it indicates a Boolean result instead of an assignment. In the programming environment I would rather formulate this as:
D |= ((A&1^1)&((B>>3)&1^1)&((B>>2)&1^1)) << 3
(sorry, also here I made have introduced some mistakes as I am not a specialist in programming)
I know that this is not evident to realize this in the current architecture, I'm just rambling some thoughts I have for the moment and ask myself what would be the best way to go about it. For the moment I integrated the holy-grail formulation in logisim-evolution.

kevinawalsh, 2019-04-15:
Sadly I don't think there is really a holy grail here, just a mess of different and sometimes-contradictory notations used in different communities.  Even the precedence rules of the various operators vary from source to source and from language to language. I'm not crazy about the bitwise operators you suggest, as it opens a pandora's box of new operators and parsing issues. Instead, in an effort to accommodate your suggestions, I've made a few smaller tweaks to the parser and renderers: 
"Engineering" notation now uses overbar (not), plus (or), dot (and), circle-plus (xor), and circle-dot (xnor).
"Programming notation" is now split into "Programming with booleans" that uses &&, ||, ==, !=, versus "programming with bits" that uses &, |, ^ ~, and ^.
"Mathematics" notation is renamed to "Logic" and "Logic Alternative" which use vee and related operators.
The parser supports all of these, and more, though the precedence rules are pretty confusing and non-intuitive.

closed 2019-04-15 by kevinawalsh