kevinawalsh/logisim-evolution #42: feature request: .BIN file support for RAM/ROM

ProxyPlayerHD, 2019-06-04:
i know it's not the most active github (sadly), but a man can dream.

title says, Binary file support. it has been suggested to the original Logisim so i'm somewhat dispointed it doesn't exist for this version of it.

the in-program HEX editor is abolsute garbage and the format it saves to it's annoying to manually convert to something workable (things like no leading 0s and x*<number> do help with file size but make it unreadable), and you know what else helps with file size? using binary files!

really the largest file you could get with the default binary format would be 64MB. (that would be a 16M x 32 RAM/ROM component) so i don't see why this isn't supported from the start.

kevinawalsh, 2019-07-05:
This is fixed in HEAD, and will appear just as soon as I can get the the 4.0.0 release out (just waiting on apple signing keys now). The new code supports a *lot* of new import/export options for RAM/ROM data, including raw binary, hex data (in the style of xxd dumps), and so on. There is a new UI when you try to import or export RAM/ROM data letting you preview and select various options. I hope you like it. Feedback is always appreciated!

closed 2019-07-05 by kevinawalsh

ProxyPlayerHD, 2019-07-06:
thanks for the reply, i will look forward to it.

sorry for the harsh words i was pissed off by the hex editor when writting this.

kevinawalsh, 2019-07-07:
No trouble. I'll let you know when I get the new binaries up.

ProxyPlayerHD, 2019-07-08:
wait i was wondering

how will this work for non-multiplies of bytes?

like if you have a ROM component that has a data width of 1 bit. will the data be expanded into the next largest amount of bytes (so, 1 byte, 2 bytes, 3, 4, 5, etc) when storing it as a .BIN file? or powers of 2 of bytes (so 1 byte, 2 bytes, 4, 8, 16, etc)?

for example:
1 bit -> 8bits (but only uses the lowest bit)
17 bits -> 32 bits (if power of 2 expanded) or 24 bits (if byte expanded)

kevinawalsh, 2019-08-30:
New binaries are up now.

There is a detailed comment here: https://github.com/kevinawalsh/logisim-evolution/blob/holycross/src/com/cburch/logisim/gui/hex/HexFile.java

To partially answer your question, you are given options for big endian or little endian (when that matters), along with lots of other options for how to convert the file data (which might be text or raw binary) into a list of words to put into the ram. For most of the formats, all of the bits are used. For example, you can set options for "binary big-endian", in which case each byte of the file is taken as a sequence of 8 bits (with the most significant bits first), all those bits are taken together as a stream of bits and used to fill up the ram words bit by bit (starting with the most significant bit of the first word, of course). The comment in the source code has more explanation.

ProxyPlayerHD, 2019-08-31:
well, good job man! now i can directly load the .BIN Files my assembler spits out.

i also like the higher speed options, it actually does make a difference in a lot of my circuits

though i will have to get used to the new placement of the icon bar at the top, with the hand, cursor, etc

the only things i can still complain about are:
1. the build in documentation is rather lacking, though i'm sure that's a already a target (saying this because i noticed a new component and have no idea what it does, the file viewer)
2. no native High DPI Monitor, it makes everything a bit blurry but not a major issue

kevinawalsh, 2019-09-01:
Regarding the icon bar at the top: You can now drag/drop those to
rearrange! You can even add other items to the toolbar up there (just drag
a component or circuit from the left side explorer/toolbox and drop it into
the toolbar at the top.

Documentation: this is the very next thing on my list. It's a big task
though, and I ran out of time before the semester started. I'm always
hopeful someone will volunteer to help out. Perhaps I might hire a student.

File viewer: You configure it with a text file. It has one input that is
used as the line number, and it will display that line of the text file. It
is meant to be used as a kind of "Assembly Viewer" for CPU-like circuits
(use the PC as the line number), for example. Or it could be used as a kind
of "unix fortune", by having a file with quotes/proverbs, one per line, and
feed a random number to the component as the line number.

high DPI: I don't have one, so can't easily test this. Any suggestions for
how the code would need to change?

ProxyPlayerHD, 2019-09-01:
with the icon bar i just meant that it was at the very left of the screen and now it's closer to the center, that just messes a bit with my muscle memory.

with the file viewer i thought you meant that you could use the actual assembly file that was used to write the program... but that is not really so easily done.
since not every Instruction only takes up 1 address it would desync the File Viewer very quickly.
for example `LD SP, **` this instruction takes 3 bytes, so 3 memory addresses, so the file viewer would have to remain on this line for the current and the 2 following addresses.
I think something easier would be to latch the instruction when the CPU reads it and use a txt file with all possible instructions inside it. for example like `0x00` would correspond to instruction `0x00`
the downside of this is that you cannot see labels or values since it only shows the actual instruction
for example `LD A, 0xFF` could only be shown as `LD A, *` since the immediate value is not part of the instruction itself.

also this reminds me, the dual and quad line RAM/ROM components are broken.
i thought i could use that feature to also have access to the immedate values so i could display them seperately.
the idea behind them is (i thought atleast) that you can access the current address and the following 4 addresses, but that is not the case most of the time.
for example if i were at Address `0x0E` with Quad line RAM/ROM i should be able to also access Address `0x0F`, `0x10`, and `0x11`. but i can access none of them since the data outputs only output Errors when the address doesn't align with a multiple of 4.

lastly, a change i really don't like, drawing wires. usually form whereever you start drawing a wire you can always draw 2 wires that are at a 90° angle to eachother... this is no longer possible IF you start draing from another wire's end.
here what i mean as picture:
[image]
and here as [VIDEO](https://i.imgur.com/PM3EQ3p.mp4)

doesn't seem like much of a change but it is, it makes wire drawing quite a lot slower because now you can only draw 1 new wire from the old on instead of 2.

kevinawalsh, 2019-09-01:
> with the icon bar i just meant that it was at the very left of the screen
> and now it's closer to the center, that just messes a bit with my muscle
> memory.
Ah, I see. Could you add a new issue to "add a 'top of window' choice in
the preferences for toolbar location", it that is what you mean? I've
currently got hidden, north (of the drawing area), south (of the drawing
area), west (of the drawing area), east (of the toolbox), and
down-the-middle (between toolbox and drawing area).

> with the file viewer i thought you meant that you could use the actual
> assembly file that was used to write the program... but that is not really
> so easily done.
> since not every Instruction only [...]
You would need to add line numbers to the file , e.g. "0x100: LD SP, **". I
considered allowing some kind of script/regexp to map the logisim input
value to file line number, but it seems overly complex.

I think something easier would be to latch the instruction when the CPU
reads it and use a txt file with all possible instructions inside it. for
example like 0x00 would correspond to instruction 0x00

> the downside of this is that you cannot see labels or values since it only
> shows the actual instruction
> for example LD A, 0xFF could only be shown as LD A, * since the immediate
> value is not part of the instruction itself.
That would work, and is a reason to keep the logisim component generic,
rather than hard-coding a PC-to-linenumber schema.

> also this reminds me, the dual and quad line RAM/ROM components are
> broken.
i thought i could use that feature to also have access to the immedate
> values so i could display them seperately.
> the idea behind them is (i thought atleast) that you can access the
> current address and the following 4 addresses, but that is not the case
> most of the [...]

Non-aligned access is not plausible in actual (simple) memory circuits.
This is something that would be better built using a dedicated circuit that
stalls the processor and fetches/reassembles the needed rows, or causes an
interrupt in the cpu and is handled, etc. There are too many variables and
approaches to hard-code a single obvious "correct" way to do non-aligned
accesses.

> lastly, a change i really don't like, drawing wires. usually form
> whereever you start drawing a wire you can always draw 2 wires that are at
> a 90° angle to eachother... this is no longer possible IF [...]
Try the space bar and backspace while drawing wires. It is a new feature, I
hope you like it. Drag a wire in a random path around the screen: each time
you press space, it puts a corner and starts a new segment; each time you
press backspace, it removes the previous corner to make a more straight
path.

ProxyPlayerHD, 2019-09-01:
well then i will have to get used to the spacebar, that is something i did not expect but will probably be very useful..

kinda sad the dual/quad port RAM don't work in non-aligned setups, but i assume they still have their use.

i think i'll just stick to the showing of Instructions without any immediate values. just seeing what instruction is being executed it already pretty useful, mostly for debugging reasons.

here a program for my current CPU that calculates the fibonacci numbers up to 6765 (sadly not higher as the BIN to BCD component is limited to 13 bits): [VIDEO](https://i.imgur.com/IYCYkjL.mp4)

ProxyPlayerHD, 2019-09-28:
one thing that came to my mind...

and i'm not sure it would be worth opening a new issue:

Carry in and Carry out pins for the Shifter.

so you can choose what bits get shifted in and what the bits are that are being shifted out.
currently i have to do this via 2 splitters.

[image]

now you might be thinking "what about shifting multiple bits?" and for that i thought the "Distance" input could simply be replaced by a regular option on the side when you select it.
then the amount of bits in Cin and Cout would change depening on by how many bits you shift at the time.

lastly... why are controlled buffers not synthesizable when trying to export as Verilog?
I mean this is what the code for a controlled buffer could look like... and it does synthesize perfectly fine, so why can't Logisim do it?

[image]

i still think i have some questions/suggestsions but for now that is enough.

BFH-ktt1, 2019-09-28:
Dear ProxyPlayerHD,

I would like to answer to your last question concerning the controlled buffer:
I agree with you that the VHDL/Verilog implementations are easy for such a component, however, there were some considerations when I started the FPGA-commander that prevented me from enabling this component:
1) Putting tri-states of several levels of hierarchy is quite some book keeping to do (complexity in the netlist generation); this was not a very important issue.
2) Many of the current FPGA/CPLD's on the market only support tri-state on top-level (meaning IOB's), hence it does not make much sense to implement them on lower level. A thing which makes more sense here is to create a specific IOB-component that is still on my todo list, but is a bit of a twisting to get it into the logisim engine. This was already a higher priority.
3) The highest priority was the fact that in the a former company where I worked I saw engineers blowing up FPGA's by improper use of these tri-state buffers. As the intention I had with the FPGA commander was educational purposes, I did not want my students to show that the interior of the nice black squares on their boards is holy smoke :-)
I hope this answers (partially) your question why logisim does not support controlled buffers.

ProxyPlayerHD, 2019-09-28:
huh, i never thought about the tristates creating difficulties

well i guess i'll have to use AND gates

or maybe there could be a controlled buffer that either pulls the output up or down when disabled?

ProxyPlayerHD, 2019-10-07:
ok now to follow up to that question, because what else can i use that puts a disabled output into a high-Z state?

because now the only controlled buffers left in my circuit are used to turn off the Address/Data bus and control lines, so other devices can run the Bus without causing shorts.

if you can't use tri-states inside the module itself then how can you have multiple devices on the same bus? like, a lot of people write emulators with these, would they have to change the entire system architecture to allow for tri-state-less use?

kevinawalsh, 2019-10-07:
Yes, that is correct. As a first-order approximation, FPGA synthesis simply
does not support shared buses (that is, a single bus with different
components driving it at different times). All signals internal to the FPGA
are uni-directional and must have a single component driving them. Instead,
to get a similar effect as a shared bus, you must rearrange the design so
that all components outputting to the bus instead feed into a single
multiplexer which drives the "bus". The multiplexer's control signal would,
in effect, ensure that only one value at a time is ever driven onto the bus.

That said, in principle, some FPGA synthesis situations can support some
forms of shared busses. I've heard of some older FPGA chips that have some
limited internal tri-state resources in the routing fabric, but my
understanding is that no modern FGPA devices have these. I've also heard of
some toolchains that will automatically convert a shared bus in a design
into a wired-OR or wired-AND bus automatically in some situations (or maybe
even insert a MUX instead if the arbitration logic is simple enough?). We
could do that in Logisim too, essentially just inserting a big OR (or AND)
of all the bus drivers for a shared bus during the pre-processing stage
when converting to HDL. But I haven't implemented that yet, since Logisim's
own support for simulating bi-directional busses is spotty at best, and
probably broken in a lot of cases (when dealing with sub-circuits for
example).

kevinawalsh, 2019-10-07:
I should also add that FPGAs do have tristates in the I/O resources, but
that's essentially only at the "edge" of the chip. Using those tristates is
currently supported in logisim using the "Port I/O" component.

ProxyPlayerHD, 2019-10-08:
you know after i made my comment and rephrased my google search a few hundred times i came across this [PDF](https://www.microsemi.com/document-portal/doc_view/130059-threestate-bibuf-an) which basically just explains how to convert a shared bus to a multiplexed one and gives some examples

a very simple to use example for a system would be this:

[image]

you any amount of devices who's outputs are connected to a multiplexer, the output of the Mux is then connected to the input of all devices. all devices with an input have an enable line which comes from a Decoder.
so overall you have a (multibit) select line to say what device is putting it's data on the "bus", and a (multibit) enable line to say what device is supposed to read in the data.

and the thing is... i've (technically) been doing this already... like if you take a look at one of my CPU's Register subcircuit... it uses a Multiplexed Design and i didn't even really realized it
[image]

so yea, i guess i can easily use this design because i've been doing it without thinking about it.

though it will be different since in the case of a full Computer sytsem on an FPGA the devices on the Bus control themself, instead of being controlled externally, which i guess can be done with simple assignments or a lookup table

kevinawalsh, 2019-10-09:
Yes, that circuit is one approach. Another approach, called "wired-OR",
uses a single N-bit wide OR gate in place of the multiplexer (plus an
enable signal for each device output so that the output will be all zeros
when not enabled). Wired AND does the same, but with an N-bit AND gate. So
long as only exactly one output device is enabled at any one time, these
are all equivalent to each other.

From a usability / user-interface standpoint, it's a lot more wiring to
draw, so it would be nice if logisim could just do the transformation
automatically when synthesizing.

ProxyPlayerHD, 2019-10-10:
Ok, I understand why the tri state buffer is not supported... but why the PLA's???

I use them so I dont have to use a bunch of regular gates, and since they can be replaced by just using normal gates i dont get why they cannot be converted to Verilog...

kevinawalsh, 2019-10-11:
Well that's just because I haven't written the verilog support for PLA yet.
Are you volunteering? I did VHDL support already though, a while ago, and I
don't expect verilog to be difficult.

I'll put it on the list of things to do.

BFH-ktt1, 2019-10-11:
Unfortunately currently more components are supported in VHDL as in Verilog (even some modules supporting Verilog have bugs and do not work correctly (e.g. Random generator and some ohters (see issues in logisim-evolution). Fortunately all synthesis tools support both Verilog and VHDL so that you can generate the HDL in VHDL. Also on my TODO list to fix these issues, however at low priority.

ProxyPlayerHD, 2019-10-12:
aww that's kinda sad. i always prefered Verilog as it doesn't look so alien to me like VHDL does.

i'm almost tempted to jus trebuild the circuit in Digital again and use that to convert it to Verilog... seems much less confusing plus it throws everything into a single file. only problem is i need to rebuild the entire thing which is just painful