kevinawalsh/logisim-evolution #45: Black screen when program is on Primary Monitor

ProxyPlayerHD, 2019-09-02:
this is a very strange one.
i was trying to use the new v4.0.0 on a laptop on work that i connected a secondary screen to.

and somehow this is happening: [VIDEO](https://i.imgur.com/N61lOeg.mp4)

info:
OS: Win7 Ultimate x86_64
CPU: i3-350M
RAM: 6GB
Java version: v8u201 (same i have on my PC at home and there Logisim works perfectly fine on both of my screens)

this is not happening with v3.1.1

ProxyPlayerHD, 2019-09-20:
i got the same problem just now on another system. this time with Win10

does it have to do with the Graphics Card?

because my main PC has an Nvidia Card and there it works fine on both screens....
but the systems i have problems with it don't have Nvidia Cards. (the laptop in the main post uses Intel Integrated Graphics while the PC i'm on now uses an AMD Graphics Card)

what is confusing me even more is that when i run it as administrator on the AMD machine it works perfectly fine...

what is going on

kevinawalsh, 2019-09-20:
That's a real mystery. I haven't had the chance to look into it yet, but I
have a few ideas of where to look in the code. I use dual monitor setups on
linux, with no trouble. I'll try on a few different machines to see if I
can reproduce the issue.

ProxyPlayerHD, 2020-01-31:
any luck yet? even with v4.0.4 i still have this issue on machines with AMD or Intel GPUs

kevinawalsh, 2020-02-03:
I still can't reproduce, and I have found nothing unusual in logisim code that could cause this issue. I have found several bug reports on it -- buggy graphics drivers, bugs in java, etc. 
Could you try some of these options:
 - Change the graphics acceleration options within Logisim (look in File :: Preferences :: Experimental)
 - Run logisim with "-Dsun.java2d.d3d=false" on the java command line. So more or less like:
    java -Dsun.java2d.d3d=false -jar Logisim-Evolution-4.0.4-hc.jar
   (doing this from within cmd.exe or powershell). Also look for errors getting printed to that command line window as well.
 - If that doesn't help, maybe one of these two...
 - Similarly, run with "-Dsun.java2d.ddoffscreen=false" option.
 - Similarly, run with "-Dsun.java2d.noddraw=true" option.

In the meantime, I have a potential workaround I will implement and test, then send to you.

kevinawalsh, 2020-02-05:
I have checked in a possible fix with f7d54fd83239ed91387750749f4e913541b6fdcd

This adds a new option to the App Preferences dialog, look under "Experimental" for the dual-screen bug fix option. There are 3 different choices. I think the first might be sufficient, but the other two options are wild guesses on my part in case the first doesn't help. This seems to be a java/swing bug related to high-dpi or 4k displays and I can't yet reproduce it.

If you want a jar, you can find it here:
http://mathcs.holycross.edu/~kwalsh/logisim-evolution-4.0.4hc-dualscreen.jar

If you want windows exe or a mac installer package, that will take more work, but please just ask.

ProxyPlayerHD, 2020-02-06:
I'll give it a try once i'm at a system again with multiple monitors (aside from my main PC)