kevinawalsh/logisim-evolution #49: feature request: wire cutters

kevinawalsh, 2019-09-15:
Some kind of wire-snip / wire-deletion tool would be nice.

ProxyPlayerHD, 2019-09-27:
???

but you can already do that with the selection tool...

[VIDEO](https://i.imgur.com/iSxfvNf.gifv)

kevinawalsh, 2019-09-27:
Yes, wires can be deleted with the selection tool. But watching students
laboriously select individual wires, press delete, then redraw the part of
the wire they wanted to keep -- its painful. They are looking for more
streamlined way to slice away parts of the circuit.

One option: a cutter tool that when you click a wire simply deletes that
wire. Or drag across multiple wires to delete them all.

Another option: When selecting a region of the screen, partial wires can be
selected and deleted. This would allow for selecting a narrow, tall
rectangle that intersects a number of wires, then delete would split all
those wires in two leaving a gap between them.

ProxyPlayerHD, 2019-09-27:
>But watching students laboriously select individual wires, press delete, then redraw the part of
the wire they wanted to keep

as said before you can just split them up with the selection tool like shown in the video, no need to delete whole wires.

>a cutter tool that when you click a wire simply deletes that
wire. Or drag across multiple wires to delete them all.

selection tool -> DEL key
exact same function without requiring an extra tool. also works across multiple componenets/wires

>partial wires can be selected and deleted

that's the only thing i would actually see being useful.
but then it should be done with an extra key, like holding SHIFT while selecting to select parts of wires, otherwise it would get very annoying very quickly.
but not sure how that would be implemented

kevinawalsh, 2020-01-09:
Added a new "Wire Cutter" tool in f8b6a605290dd4f2d396ce9d6f7497edb2c270f0.
Having this be part of the current Edit/Multi-tool was attempted (see d1c48ba18a68713483182335dfc467d3582c65a8) but just had too many unresolved UI issues, so this dedicated wire-removal tool was added instead.

closed 2020-01-09 by kevinawalsh