kevinawalsh/logisim-evolution #50: large 4-input OR gate looks funny

kevinawalsh, 2019-09-29:
One of the pins is dangling in a strange way.

BFH-ktt1, 2019-09-29:
I observed that it is the case for all gates in large form with 4-inputs, jfyi

BFH-ktt1, 2019-09-30:
Can be fixed by adding an offset in AbstractGate.java (line 270 .. 277):
old:
    int dy;
    if ((inputs & 1) == 1) {
      dy = skipStart * (inputs - 1) + skipDist * index;
    } else {
      dy = skipStart * inputs + skipDist * index;
      if (index >= inputs / 2) dy += skipLowerEven;
    }

new:

    int dy;
    if ((inputs & 1) == 1) {
      dy = skipStart * (inputs - 1) + skipDist * index;
    } else {
      dy = skipStart * inputs + skipDist * index;
      if (index >= inputs / 2) dy += skipLowerEven;
      if (inputs == 4 && size >= 60) dy -= 10;
    }

kevinawalsh, 2020-01-08:
Fixed in 472a75e3e0578c98f1bda6b1a6d425b509b28119, with a warning upon file open about this backwards-incompatible change.

closed 2020-01-08 by kevinawalsh