kevinawalsh/logisim-evolution #54: feature request: Built-in VHDL simulator

TChapman500, 2019-10-23:
I think that Logisim needs its own built-in VHDL simulator.  It doesn't have to be too advanced.  Just enough to propagate signals.

kevinawalsh, 2019-10-24:
I am not a VHDL expert. But what little I do know about VHDL makes me think
that even just parsing VHDL is non-trivial, never mind simulating it. If
there are open-source VHDL simulators that can be bundled and integrated
with logisim, on the other hand, that seems like a reasonable approach.

BFH-ktt1, 2019-10-24:
Currently the most promising open-source VHDL simulator is GHDL (http://ghdl.free.fr/; https://ghdl.readthedocs.io/en/latest/about.html). It is still on my TODO list to see how it can be integrated into/used with logisim.