kevinawalsh/logisim-evolution #57: weird visual glitch with larger circuits [4.0.1]

ProxyPlayerHD, 2019-12-16:
usually all components have gray pins when not connected and when using the poke tool you can look at the state of a wire...

well after a certain distance from the top left of the circuit this kinda stops working.

components get black/blue pins (black if pin is wider than 1 bit) and using the poke tool doesn't seem to work anymore. even when the wire is showing a high or low state the poke tool still only shows "X"s but only after a split point in a wire (the split points are always blue on single bit wires)

it's hard to explain, so here a few screenshots.
how it's supposed to be:
[image]
how it looks if far enough away:
[image]
both right next to eachother.
[image]

no idea why this is happening, i also tried it on a completely new file and it still happens. so this should be easy to reproduce

kevinawalsh, 2019-12-16:
Nice catch. This is almost certainly a fault with the "fast path grid" in CircuitState. State for any point within 2000 units of the origin is handled using a dense 2d array. This makes most simulation faster, at the cost of increased memory usage. Anything further away uses a slow path hashtable, trading off speed for reduced memory usage. That code was a quick hack to improve performance, and there must be a regression in the drawing code where the slow path is failing.

kevinawalsh, 2020-01-07:
Fixed by c1337692098bb8c90247b49f2263182379d1b202

closed 2020-01-07 by kevinawalsh