kevinawalsh/logisim-evolution #58: feature request:  Logic Element Estimator

TChapman500, 2019-12-21:
Add something that gives an estimate on how many FPGA logic elements will be used by a given circuit.  For example: "FPGA -> Logic Block Estimate".

kevinawalsh, 2019-12-22:
Great idea! But is this even feasible? Logisim already has a built-in
stats/reporting tool that will show you some simple statistics about your
design (number of circuit elements -- gates, etc., -- broken down by type,
etc.) But for an FPGA, "logic elements" is a complex and fuzzy concept. It
would likely depend on vendor, device, etc., and fairly complex rules. It
would probably also depend on the pin I/O resource assignments. Can this
can be farmed out to an external tool?

TChapman500, 2019-12-24:
Good points.

kevinawalsh, 2020-01-08:
Closing as wontfix for now. Will re-open if new ideas come up.

closed 2020-01-08 by kevinawalsh