kevinawalsh/logisim-evolution #59: Bug:  Adjust Transmission Gate Behavior

TChapman500, 2020-01-13:
Currently, the transmission gate outputs an error signal if one half of the gate is on while the other half is off.  It should be that if the gate is half on and half off, then the output is whatever is on the input.

kevinawalsh, 2020-01-19:
I made some tweaks to the transistor behavior in 9b24d5b729b0a525ea4cbbb6d65848ce989f40a2, but I can't reproduce this specific "transmission gate outputs and error signal" behavior. As far as I can tell, the transmission gate is working as intended.

Note that the intended (and now documented) behavior of the transistors and transmission gate are not even remotely close to reality. They just give a vague impression of what a transistor and transmission gate. The transmission gate in particular is *not* bidirectional. It can only propagate signals in one direction (as indicated by the arrow). That is due to fundamental limitations of Logisim's circuit simulator, which was never designed for analog simulation and really can't handle bidirectional signals well at all either (splitters are true bidirectional objects but they are a special case and are hard-coded deep into the simulation engine, and the few other bidirectional things like certain memory components are very constrained in how they can behave).

Nevertheless, if you can give me a test case or even a video or screenshot showing this transmission gate error, I will reopen this and try to reproduce it.

TChapman500, 2020-01-19:
This is what I'm talking about.  Not trying to do any bidirectional stuff with the transistors.  As you can see here, one transistor is on while the other is off.  The result should be that the data is forwarded to the output in this case.
[image]

kevinawalsh, 2020-01-19:
Those error outputs are intentional (and documented in the library reference). The upper and lower inputs (pgate and ngate) are normally intended to be inverses of each other. Make the top one a 1 and bottom one a 0, for example, or vice versa. Or stick a not gate on the wire leading to the bottom pin. That data doesn't get forwarded when "one transistor is on and the other is off". It gets forwarded when both transistors are enabled (meaning the one has a high input and other has a low input).

I notice now that the mouseover text for the two pins is incorrect, however, and I'll fix that shortly.

Really this whole transmission gate (and transistors) are probably best just removed from logisim entirely. I've moved them to a separate "Analog Library" to avoid accidental use by students. There is a "Tri-State buffer" (aka "Controlled Buffer") if you just want a way to pass data through when the enable is 1 and disconnect the output (so it is floating) when the enable is 0 That component has just a data input and an enable input..

TChapman500, 2020-01-19:
I discovered this issue when I tried making a multiplexer out of these, and then latching the output back to the input.  The inverter causes the conditions you see in the image.  Just enough to make multiplexer-based latches using these things impossible.  Also, I was not aware that there was an analog library.  Using the independent transistor components gives the top image a floating result when Data is 0, and the bottom image a floating result when Data is 1.

kevinawalsh, 2020-01-20:
The analog library is new. It just contains the few non-digital components that I prefer my students not to use (transistors, pull resistors, power/ground, and transmission gate), which used to be under "wiring".

You can make a mux out of transmission gates if you just put NOT gate on the upper (or lower) input to the transmission gate. The upper and lower inputs need to be opposites, otherwise it just outputs an error. Or you can use tristate buffers to do what you are aiming for.  

The behavior you describe for the transistors is intended -- that is how CMOS transistors are meant to work. The Ptype and NType are used in different ways, one of them only for pulling down, the other only for pulling up.

Note that *none* of this stuff can be synthesized for fpgas, since those really do not support tristate devices or transistors at all.

closed 2020-01-20 by kevinawalsh