kevinawalsh/logisim-evolution #61: Feature Request: Color control for High/Low signals

ProxyPlayerHD, 2020-01-24:
the idea is simple. allow people to set custom High and Low colors for the simulator.
i came to this because i used Logisim Evo to make a logic circuit that i showed online and someone had problems seeing the state of the wires because of how similar the green tones are. so being able to set custom colors would definitely help color blind people. would also be useful if this were saved for the whole program, so when you completely close it and open it again it still has the same colors.

couldn't you "just" use the same color picker for the color of LEDs, the Terminal, etc?

kevinawalsh, 2020-01-24:
This should be easy. It could be:
 (a) a global setting for logisim (these are stored in hidden settings file the user's home directory), which would apply to all circuits opened with that copy of logisim
 (b) a setting within each ".circ" file, which would apply to all of the circuits and subcircuits within that file (but not to imported ".circ" libraries, which would have their own color settings)
 (c) a setting within each circuit, so different circuits in the same file could have different colors.

I would think (a) global setting would be the most user-friendly choice, since you are bringing this up as an accessibility issue (which is tied to a user) not an idea for customizing circuits or files.

Are there other things besides wire colors (active, inactive, bus colors, etc.) that you think should be customizable?  Maybe the canvas background color... it is currently just white, but maybe light gray/tan or even a dark mode would be nice for some people?

Kevin

ProxyPlayerHD, 2020-01-24:
well the default 4 wire colors. 0, 1, X, E

then also the background, and maybe one for the insides of Gates? (so by default it's transparent but maybe they could be set to be filled with a color, but then the text ontop could get hard to read so i don't know really)

the background would definitely be useful, not just for accessibility. i had it multiple times that the program's white background almost burned my eyes out when i opened it in the middle of the night.

ProxyPlayerHD, 2020-03-27:
how is the progress going on this?

kevinawalsh, 2020-03-27:
Quarantined?
The color code is entirely done, and looks great. But it required going
through many (well, all, actually) components to fix graphics glitches.
Nearly all components were written with the expectation of the standard
color scheme.
I'm swamped with transitioning my courses to an online format, and dealing
with quarantine issues.
Just as soon as I get some free time I'll try to finish this up and push it
out.
Kevin

ProxyPlayerHD, 2020-03-27:
oh i see, i thought it would be exactly the opposite, that you would have a lot more time due to the Quarantine.
atleast it's like that for me.
stay healthy, and good luck!

kevinawalsh, 2020-03-27:
Sadly, no. We are still teaching full time, but now have illnesses,
inconsistent timezones, buggy software, stressed students, and children at
home full time to cope with.
Say safe and healthy yourself!