kevinawalsh/logisim-evolution #68: Minor bug

John-Lluch, 2020-09-05:
Hi, I have tested your fork in a quite large project and I found with great satisfaction that your version feels and **is** _much_ faster than any of the previous forks, particularly on simulation. I also like the additional user interface options and simulation feedback that the app provides, and I think this is overall a better work than any of the previous ones!

I found a minor user interface glitch. After a circuit has run for some time but then it suddenly pauses due to a "oscilation apparent" issue, the Auto-Propagate menu option remains checked despite the simulation was paused and the Single-Step Propagation is available.

kevinawalsh, 2020-09-06:
Thanks for the kind words. I put in a bit of time and effort on the simulation aspect, so that we could simulate full 32-bit MIPS / RISC processors at a reasonable speed, or do real-time audio generation. It's nice that someone noticed!

I'll take a look at the glitch. I've had trouble with that piece of code before, so I'm not surprised it is buggy.

John-Lluch, 2020-09-08:
> It's nice that someone noticed!

Well, your app is definitely much faster than anything else, by a very long extend, and provides a much better experience for very big projects. It's kind of surprising to me that this is not getting a lot more 'stars', honestly.

> I'll take a look at the glitch. I've had trouble with that piece of code before, so I'm not surprised it is buggy.

That's great. In fact, there's some more oddities on the behaviour of the 'auto-propagate' feature. Another problem that I found sometimes is that Auto-Propagate can not even be disabled after "Oscilation apparent" happens, unless I Reset Simulation. At the same time, it looks that the Single Step Propagation is always available, which makes the whole thing a bit confusing. Not sure if this is intended for some purpose, but I found that on the other forks, the Single Step Propagation is only available when the Auto-propagate is off, so it's either 'Auto' or 'Manual'

Another glitch that I found is that the normal purple squares that show when changing values while in simulation, do not disappear after going back to Editing mode. Purple squares may even stay around while editing without going away. See this screen-shot that I created after setting the input to "1", then going back to Edit, and moving the input to a different location.
[image]

Other than that, the app is just great so far. I even noticed that you changed the behaviour of transistors in a way that is more coherent with what I would have expected, so that's great too.

kevinawalsh, 2020-09-08:
Thanks for the feedback! I've seen the highlighting issue as well, but it has been low priority and I haven't had a chance to work on that. In fact, with the covid disruptions, I likely won't get to work on Logisim at all for a bit except for critical fixes.

John-Lluch, 2020-09-08:
There’s no need to rush for that, really. It’s already quite significant that people like you is using their time to care about this old software and bring improvements. So I can just say thank you for that!