kevinawalsh/logisim-evolution #69: Feature Request:  Option to Revert to Wire Placement Old Behavior

TChapman500, 2020-09-08:
I really do not like the new wire placement behavior and would like to have the option of reverting to the old one.

kevinawalsh, 2020-09-08:
What is it about the new placement behavior that you don't like? I did make some changes around da36abb5481f5256e37158bdd2001125bde67cd2, 13cff8843ed66f3f327495938cd30e57107968f2, and e34cce0ccb2e0fa4ad28da59a70789dc48a03e69, but I've been using the new version a lot so I don't even recall what the old behavior was. Can you explain what you are looking for? I'm hesitant to simply revert, since I must have had a reason for those changes.

TChapman500, 2020-09-08:
In 3.1.1, the wire placement behavior was more predictable when appending to existing wires.  The old system did not move the existing wires when appending a new wire from it (the main problem with the new system).  Now I need to spend more time making sure that I don't move the old wires when appending to new wires or it'll break the circuit.

kevinawalsh, 2020-09-08:
Thanks. To clarify, when you say "move the existing wire", do you mean "remove"? Because in the current version, when grabbing the end of an existing wire, it will either add new segments to the already existing part, or it will remove part of the old wire, or both. But it will never "move" an existing wire.

TChapman500, 2020-09-08:
That's a different issue, but it does appear to be removing wires under certain conditions when I'm trying to extend them, like if I accidentally drag in the direction of a wire after clicking on a wire.  That breaks very deliberate connections that I made.  What I'm primarily complaining about is the fact that if I try to append a perpendicular wire to a wire end (dead end), any parallel movement of the mouse will move the previous wire end to a different location.  And that causes that wire to connect to things it's not supposed to connect to.

John-Lluch, 2020-09-26:
Hi @TChapman500 . I'm not sure if I understand you fully, but I actually like it more as it is now. In fact, I actually find it more predictable, wires just always end where you drag the mouse pointer. Also, it was not previously possible to spit a wire by just grabbing it from the middle, now it is.

TChapman500, 2020-09-26:
I hate the split feature.  It'll break a wire I'm trying to extend to somewhere else.