kevinawalsh/logisim-evolution #70: Serious BUG on Combinational analysis

John-Lluch, 2020-10-03:
I've been able to narrow down a critical bug in the Combinational Analysis tool.

It can be reproduced like this:

(1)  Open the attached CombinationalAnalysisBug.circ [CombinationalAnalysisBug.circ.zip](https://github.com/kevinawalsh/logisim-evolution/files/5322181/CombinationalAnalysisBug.circ.zip) or manually reproduce the circuit below:

[image]

(2)  Click on Project->Analyze Circuit

The circuit is correctly analysed and translated into the following logic expression: 

`x = a | ~b | c & d | e | f | g | h`

(3)  In Combinational Analysis click on "Build Circuit"

The Expression above is converted into this: 
[image]

BUG DESCRIPTION: The wire from input 'b' is misplaced near the connection with the 'or' gate, overlapping other wires and resulting in an undesired connection with the output of the 'and' gate. It appears that the 'b' wire gets moved to the left due to the negated input of the or gate. This causes errors and incorrect behaviour of the circuit.

EXPECTED BEHAVIOUR: The wire displacement to the left described above is very reproducible. Unfortunately, in can overlap nearby wires and cause wrong connections. Wire placement after "Build Circuit" should take negated 'or' inputs into account and adjust accordingly to avoid such wire overlaps

COMMENTS: The bug can be reproduced in an easier way by just copying and pasting the expression (2) in the Combinational Analysis tool.

kevinawalsh, 2020-10-04:
Yup, that's pretty bad, and I can confirm it. And thank you for the detailed report.
I just committed a quick and dirty fix for this in 1fa51fd and it does fix the case you identified. I haven't tested it extensively yet though. My time right now is very limited, so I don't imagine I'll be able to do more testing very soon, and the next release isn't likely very soon either unfortunately. But you can check out the latest code and compile yourself, or I can send you a jar file.

Thanks again,
kevin

closed 2020-10-04 by kevinawalsh

John-Lluch, 2020-10-05:
Hi Kevin
I will be happy to test your fix with the table which originally caused the issue, but I am not used to java and I do not have any compiler tools to compile the source code. I would appreciate if you could send me a .jar file instead. You will find my email address in my profile. I will report whether I consider it fixed. Thank you very much.

kevinawalsh, 2020-10-05:
Sent!

kevinawalsh, 2020-10-05:
email isn't working. Here is the file for you, or anyone who wants to try it:
[logisim-evolution-4.0.4hc-buildcircuit.jar](http://mathcs.holycross.edu/~kwalsh/logisim-evolution-4.0.4hc-buildcircuit.jar)

John-Lluch, 2020-10-05:
Hi Kevin. Thanks for that.

 I just checked and the my email is correctly entered in github, in fact I receive github email notifications just fine. 

Anyway, I downloaded the .jar file from your previous message but it does not open. The following message appears: "the Java JAR file "logisim-evolution-4.0.4hc-buildcircuit.jar" could not be launched. Check the Console for possible error messages". [But there's not any related message on the console].

 I'm on Mac OS X 10.12. The Java Control panel states "Your System has the recommended version of Java. Java 8 Update 221"  

Any ideas?

Thanks