kevinawalsh/logisim-evolution #71: Feature Request:  Viewport Panning/Zooming Overhaul

TChapman500, 2020-11-05:
The current system for zooming and panning is very cumbersome on large circuits, so I propose the following system be put in its place.

1.  Remove the scrollbars from the viewport.  Instead, the viewport should be panned using the middle mouse button.
2.  Replace the zoom in/out buttons with the mouse wheel.  Scrolling up zooms in, scrolling down zooms out.
3.  Allow component positions to be negative.
4.  Create a system (eg: different colored dots) for identifying the location of the origin and axis.

kevinawalsh, 2020-11-06:
Thanks, these are good suggestions. 
1. You can already customize the actions for each mouse button (right-click the toolbar and select "Customize"). There is no "pan/zoom" tool though. I could add a pan (or pan/zoom) tool, so people could assign that to the middle button if they want. Would that be enough? Also,  I'm not crazy about removing scrollbars. Some people use trackpads with no obvious middle buttons, I personally am old school and like scrollbars. I could possibly add an option to remove them I guess. I should also try support for two-finger (or is the 3-finger?) panning for macs or windows.
2. mouse-wheel-zoom-pan is already implemented. It's probably not documented, and maybe not obvious though.  
  * Wheel = pan scroll up/down
  * Shift + wheel = pan scroll left/right
  * Ctrl + wheel = zoom in/out
3. This would be a breaking change for the file format, which I'm a little hesitant to do. It would mean new files with negative coordinates would likely crash older versions. What's the motivation for this? Would it be enough to have it automatically shift the coordinate system and provide the illusion of infinite canvas in all directions?
4. Is there any significance to the axis and origin? I know the raw coordinates are visible in a few places in the UI, but the coordinates are currently mostly an implementation detail that users don't actually see. Is it that you just want a bounding box? Or a grid-like layout with an axis?

TChapman500, 2020-11-06:
1. I know that the wheel currently pans up/down.  I'm used to editors that center the viewport on 0, 0.  It's very cumbersome to have to move large circuits down and to the right because we can't have negative part coordinates and have to incrementally move parts to their target locations.  Mac mice have a middle mouse button and a scroll ball that (in theory) supports horizontal and vertical scrolling.
2.  Mouse wheel movement should zoom in/out, centered on the camera.  Mouse wheel press should allow user to drag the viewport around (anchoring a specific point of the viewport to the mouse while the middle button is down).  I believe I have some sample code that I could email to you (C++)  An option for the user to switch to this system or keep the current system is fine.  I never found any documentation or anything suggesting that I needed to use the modifier keys to change the wheel behavior.
3.  Shifting the circuit would be a suitable alternative, but I think that the shift should only be done when saving the circuit as that could cause a lot of performance issues while editing.  As long as we can pan the camera to give us as much working space anywhere in the viewport that we want.
4.  The axis and origin markings that I'm proposing would only be for the user to find 0, 0 if negative coordinates are allowed and the scrollbar is removed.  I have [a video](https://www.youtube.com/watch?v=FGdpPojnLOE) of my track editor utilizing some of the same technologies that I'm requesting here.  The grid system is different but I believe the concept to be workable with dots instead of lines.

kevinawalsh, 2020-11-06:
Hi TChapman -
Thanks for the clarifications.
For Mac, I don't actually do much testing on there, and I haven't regularly used a mac in decades, so I apologize if Logisim doesn't always match the mac-specific behaviors you might be used to. It would be very difficult for me to try and maintain multiple different interfaces for different platforms, so I try to pick a single reasonable behavior that works reasonably well on supported platforms (Linux, Mac, Windows).

For zooming, Logisim matches the behavior of many other programs like gimp, chrome, firefox, various text editors (word, libreoffice, etc.). Control-scrollwheel should do exactly what you are suggesting: it zooms inwards or outwards while staying centered on the point under the cursor. Regular scrollwheel or Shift-scrollwheel are for panning. This might not be common on Mac, I suppose. And yes, regrettably the documentation hasn't kept up, so these features are not very discoverable.

For moving circuits, I'm hoping you simply do select-all then move the circuit over in one motion. That's what I usually do, and I've never found it very cumbersome. Not much different from editing in powerpoint, or drawing programs, or anything else with a fixed size workspace.

I should be able to add support for the mac scroll-ball. And some basic touchpad gestures maybe. It won't be soon, due to the backlog of fixes I need to get finished, and the busy semester.

TChapman500, 2020-11-06:
I use almost exclusively Windows, but when I use the mac, I use it with a traditional PC mouse.  I was just stating that the mac wheel (more like a ball) supports horizontal and vertical scrolling.  It doesn't have anything to do with any specific "mac behavior" that I'm "used to".  What I am used to doing is working with 3D modeling software (eg: 3DS Max, gmax) that uses the middle mouse button to pan the viewport with mouse movement and the mouse wheel to zoom in and out.  The change I'm requesting is not platform-specific.
I do select all to move everything down and right, but I sometimes have to do this many times.  Some times, I only need to move a small portion of the circuit, but I can't move it far enough in one go.  This is where most of the headache comes in.  When small portions of a circuit need to be moved great distances and there just isn't a good way to do that in one shot.

kevinawalsh, 2020-11-06:
Ah right, I hadn't been thinking of 3d modeling, where the zoom/pan behavior is a bit different. I'll have to experiment with that to see how it feels in Logisim, and maybe do a quick survey of other systems to see what is more typical for 2d layout software.

Sorry this won't likely be ready any time soon, as I am swamped with other responsibilities these days.

TChapman500, 2020-11-07:
In 3D modeling, the viewports (both 2D and 3D) usually work the way I described or in some way very similar.  Wheel is always zoom, middle button is always pan (except for blender, where it's rotate).

John-Lluch, 2020-11-13:
Just to clarify what's the behaviour on a Mac computer, the zoom/pan feature using control/shift with the mouse scroll wheel is just a standard feature that works in all apps. Unfortunately, this is not what (I think) TChapman500 is suggesting. The zooming is a whole screen zooming meant for people with sight difficulties, not a canvas zooming. So it can't be used as a replacement or alternative to the zoom scrollbar.

TChapman500, 2020-11-13:
What I'm suggesting is that the **viewport** zooms in and out with mouse wheel movement (zoom center is the cursor position) without the use of any modifier keys.

kevinawalsh, 2020-11-14:
John - thanks for that tip. So control/shift mouse wheel zooms and scrolls the entire desktop on Mac? Is that for all Macs, or just if you turn on some kind of disability/accessibility settings? That's really unfortunately, since it would mean Logisim can't use those combinations on Mac. How does something like gimp, or chrome, or firefox, or most other programs handle scrolling and zooming on mac?

TChampan500 - I really don't like the idea of mouse wheel being zoom. Aside from 3d modeling programs, it seems vastly more common for mouse wheel to scroll up/down, not zoom in/out.

John-Lluch, 2020-11-14:
Hi Kevin, Yes, that's right. The zoom happened on the entire screen, with loss of resolution as you zoom more and more. It's just a system level screen zoom thing I guess, it has no interaction with apps or app code. 

In the "Accessibility" panel on System Preferences there are several ways you can configure the particular behaviour. Incidentally, I have just played with that I found that disabling "Use scroll gesture with modifies keys to zoom" it now plays nicely with Logisim (!) by doing the same as the zoom scrool bar while keeping the canvas centered on the mouse pointer. It turns to be that the default "modifier key to zoom" on the accessibility panel is the control-key; when enabled, the zoom happens on the entire screen as described earlier. So I think your implementation is right, and works just fine !

I agree with the mouse wheel acting as up/down scroll alone. It is true that some 2D cad programs, including circuit layout and pcb design software use the mouse wheel to zoom, but these programs do not adhere anyway to the normally accepted GUI rules of Mac and Windows as far as selecting, duplicating, editing items and so on, is concerned. I think that the logisim way of editing items fits more with the standard guidelines for desktop applications, and in my opinion the mouse wheel behaviour is right as it is.

On the other hand, maybe it would be great to have "navigation" button arrows that would allow you to go back and forward to circuits in the same way that web browsers do. While simulating or editing complex hierarchical circuits, the need to click down to a sub circuit to check something, just to return to the previous circuit is very common. Currently, you need to explicitly click on the interesting circuit every time. It would be nice that we could navigate to the previous circuit by just clicking on a "back" arrow.