kevinawalsh/logisim-evolution #74: Performance Considerations

John-Lluch, 2020-11-17:
Hi, there's little doubt that raw simulation performance of your fork is much better than any other fork. Still, I noticed that performance degrades _sharply_ and significantly after I reached a certain circuit complexity. Execution becomes jerky with long and inconsistent execution stalls between clock ticks.

The rather inconsistent flow of the execution, which sometimes takes very long to complete one particular step, and sometimes the same step is completed very fast, as well as the fact that this performance degradation begins to happen sharply after some project complexity, makes me think that this is related to memory constraints or to the excessive operation of the Java garbage collector.  

On the other hand, I am running Logisim in a Mac with plenty of physical RAM, 12 Gb, and Logisim rarely uses more than 2 Gb with still plenty of free RAM available while Logisim is running. So physical memory constraints do not seem to be the case... which leaves me with the idea that there's something related with the Java virtual machine that causes this, or maybe some artificial limit put in the Logisim code (?) 

What could degrade performance in such a sharp way?, does the above makes any sense? If so, is there a way to improve on that?

Thanks.

kevinawalsh, 2020-11-17:
Hi John,
The memory/gc issue could certainly be a cause. Or just a bug in Logisim (there are many). If you have a specific circuit in mind, I'm always happy to use it as a test case when I have a chance to look at the performance issues again. This may be a while, as I am swamped with other responsibilities currently.

One thing you might try is running the JVM with options to increase the amount of available memory (The JVM artificially limits the available memory, to force GC to run). You could try, for example:
```
java -Xmx4G -jar logisim-evolution-4.0.4hc.jar
```
This should allow Java to go up to 4GB. I'd be interested to hear if that helps. It is possible for Logisim to make this change internally, if I can devise a reliable way to detect that memory is causing simulation issues.

TChapman500, 2020-11-17:
The command line doesn't seem to be working (Windows 10).

John-Lluch, 2020-11-17:
Hi Kevin, thanks for your answer. I was able to run your command line after downloading Oracle jdk 15.0.1 , (not sure why I needed that) but I found there's no difference. I also tried replacing  '-Xmx4G' by '-Xmx8G' just in case, and there was no difference either.

kevinawalsh, 2020-11-17:
Ah well, that's unfortunate. It could still be memory-related, but it is sounding like perhaps some performance glitch or even bug in Logisim. If you are willing to share an example circuit (which I am happy to keep confidential), or you'd like me to tell you how to run a profiler and dig into the Logisim code, either one would work for me.

TChapman500, 2020-11-17:
Didn't I give you a large circuit file a while back where everything was discrete logic gates with no built-in components?  If you still have that circuit, you can use that.  I don't have that circuit anymore.

commit aba7d13e (2016-02-22): Bugfix fixes #74