kevinawalsh/logisim-evolution #80: Request bidirectional pins

jsbaboon, 2021-05-03:
Subcircuits with bypassing wires branching off require 2 way pins.  That is circuits can then be placed sequentially along a line for example. Thanks

TChapman500, 2021-05-04:
This has been requested many times before by various people (including me).  Unfortunately, it also seems to be the hardest to implement.

kevinawalsh, 2021-05-04:
Yes @jsbaboon, @TChapman500 is right -- the simulator engine is quite complex, and because the bidirectional pin feature was not part of the original design, the entire engine and all the data structures and pre-processing steps (which are needed to make the simulation efficient) assume that every sub-circuit pin has a single direction. Every time I have looked at trying to implement bidirectional pins for subcircuits, I have concluded that it just isn't feasible in the current simulation engine design without a substantial rewrite.

More lately, an additional issue has cropped up involving the FPGA toolchain features from REDS-HEIG. FPGA synthesis relies on Verilog and VHDL and intermediate steps, and neither of them support bidirectional subcircuit pins for simulation either.

I would still like to find a way to support this, even if it means a complete simulation engine overhaul. But I won't have time to do it anytime soon. I'd be happy to take contributions though!

closed 2021-05-04 by kevinawalsh

TChapman500, 2021-05-04:
I think you're a bit too quick to close this issue.  I have an idea on how to reduce the workload for making the implementation.  Set the internal direction to whatever is required to propagate a signal (so it's only an input or output, but not truly bidirectional).  From there, the current pin behavior should take-over.  If both sides are floating, but one side changes to be non-floating, check to see if the internal direction needs to change and change it accordingly.  If there's an error on one side of the pin, then the opposite side either (a) goes floating or (b) changes state, then flip the pin's direction accordingly and stop driving other other side.

TChapman500, 2021-05-04:
Also, the lack of support from FPGA synthesis should not be a block to implementing this feature.

kevinawalsh, 2021-05-05:
Thanks for this. That's a good point about FPGA synthesis. I was rather thinking that the lack of support in industry standard HDL simulation tools was indicative of how tricky the bidirectional pin problem really is.

As for the proposed approach, it just does not apply to the way the simulation engine works in Logisim. It does not process signals on a case-by-case basis like this. A preprocessing stage determines the connectivity and dependencies of all networks, but confined only within a single circuit, and stores this information in various interlinked datastructures across multiple levels of the circuit hierarchy. Any change to a pin direction anywhere in the entire design invalidates potentially the entire preprocessed data structure, and wipes out a lot of the simulation state and signal information. Some best-effort heuristics are implemented in the engine to preserve some state across such changes, so that you can edit one part of the circuit without disturbing unrelated signals in other parts of the design. But changing a pin direction would, at minimum, wipe out the simulation state for the entire connected network. The data structures and dependency graphs were simply not designed with the idea that pins could be bidirectional or that pins could change direction from one step to the next.

There is essentially a single non-wire "component" in logisim that is truly bidirectional: splitters. And it is because they are bidirectional that the behavior of splitters is hard-coded and deeply integrated into the simulation engine, and lots of the engine logic is intertwined with coping with splitters. Essentially, the simulation engine precompiles the design into an equivalent form that has no splitters as the first step of preprocessing.

I'd be more than happy for others to look closely at the code in question. It's all available.