kevinawalsh/logisim-evolution #83: Request a New Feature

livan3li, 2021-05-19:
hi, i've lost my data two or there times because of forgetting saving my project. I wish there was a autosave feature. if  possible please can you add this?

kevinawalsh, 2021-05-20:
Thanks, good idea.

Are you asking for a recovery feature? Similar, for example, to microsoft office or emacs/vim has so that when they crash, upon reopening they will prompt to recover files from the temporary save files.

Or, are you asking for a continuous auto-save feature, more like google docs, such that the circuit file would be automatically saved, e.g. every few minutes or whenever it changes.

Either one is probably feasible (the recovery/backup style would take some careful design for a decent user interface, and the continuous auto-save would probably need a global preference and careful tuning to avoid too much write activity. For the auto-save style, I'd maybe add a preference (default: on?) because I personally dislike auto-save feature. I often open projects, play with them a bit, then deliberately do NOT save them so that I can preserve the original as it was. This is useful when I grade student projects, for example.

kevinawalsh, 2023-07-20:
Commit c57e32d8 implements an auto-backup feature, where logisim now makes backup copies of projects as they are being edited, and will prompt to recover them the next time it opens if the file wasn't properly saved or logisim crashed, etc.

This is not exactly "auto-save", which I've decided not to implement for now for pedagogical reasons. I like people to understand the concept of a "file".

closed 2023-07-20 by kevinawalsh