kevinawalsh/logisim-evolution #85: Minor ui issue

nailed-barnacle, 2021-06-11:
When attempting to edit the contents of the RAM object, moving between ram cells using the arrow keys nudges the entire drawing sideways and up and down.

To reproduce, in a new project, select a default RAM object and place on the work space. Using the pokey-finger (I'm sure there's a technical name for that!) select a cell to edit. Use the arrow keys to move to other cells: in most cases the highlight remains in the same place and the selected cell is changed by the RAM object moving under it.
There is an oddness which I can't fully quantify yet whereby in some cases the RAM object does not move and the highlight moves as expected; something to do with whether the highlight has moved to a previous row, I think, but I'm not yet certain.

It's a something or nothing issue, but it puzzles me I feel it shouldn't do it! Or is this expected behaviour?

Sadly I have no knowledge whatsoever regarding Java or I'd be happy to plough into it.

Neil

nailed-barnacle, 2021-06-11:
Forgot to mention: version 3.4.5

Neil

kevinawalsh, 2021-06-12:
Thank for the bug Neil. I can reproduce, confirm the bug, and I know where in the code the fix needs to be made. There are a smattering of similar bugs with other components. It happens when components don't fully capture the focus allowing some keypresses to propagate up to the main circuit canvas. Maybe the arrow keys should not be used for scrolling at all, since they are used for so many other things (like editing / "poking" RAM, rotating components, etc.).

BTW - I call it the "poke tool", but "pokey-finger" works for me!