kevinawalsh/logisim-evolution #86: Feature Requests: Subfiles and unlimited truth tables in analysis

jsbaboon, 2021-06-12:
I want to group similar components in projects with lots of components in the file window.
Or make it easier to float a file to the right position from the bottom of the list.
Or make the new files placed at current point in list.

My projects have more than the limited number of input/output pins for analysis, I dont care if the truth table is big.

ps. Your wire routing algorithm looks unsystematic and untidy when creating circuits.

kevinawalsh, 2021-06-13:
Can you clarify what you mean by "the file window"? Is this the side panel showing the libraries of components that can be added to a circuit? If so, there are a few things you can already do (but not well documented) that might partially address your feature request: 

* You can drag/drop components into or out of the toolbar, and arrange them in any order you like.
* Within a project, you can drag/drop subcircuits so they are arranged in any order you like.
* You can make a project (say, MyComponents.circ) with some collection of subcircuits, then within another project you can add this as a "Logisim Library". It will show up as a folder in the components panel, and you can use those components just as you would any other built-in component. If you have lots of components, you could make several such libraries, and add them all to your main project file.

For "float to the right position" idea, are you talking about rearranging the subcircuits listed within a project? If so, you should be able to just drag/drop the subcircuits to rearange them. The blue up/down arrows at the top work, but I agree are clumsy.

For analysis, how many inputs do your circuits have? With 20 bits of input, there are over a million combinations that need to be analyzed, and the way the analysis code was written makes this extremely slow. No one has ever asked for more bits before. How many do you need?

I agree the routing algorithm for the "build circuit" feature can be a bit ugly. It follows a simple algorithm for layout.

closed 2023-07-07 by kevinawalsh

jsbaboon, 2023-11-23:
Thanks for replying. It was a long time ago, I dont recall right now. I am
glad you made version 5 now. I dont check here often. Thanks again.