kevinawalsh/logisim-evolution #87: Feature Request:  Clear Color for "Input/Output -> RGB Video"

TChapman500, 2021-06-28:
For the RGB Video component, I think it would be good to have several options for clearing the display (set by the properties window).

1. Fixed Clear Color
2. Input

Upon receiving an clear signal, the screen would become the color selected either by the input or by the fixed clear color property.  If the user selects "Input", but provides a floating or error signal, or there is no connection, then the Fixed Clear Color is used.

BFH-ktt1, 2021-09-17:
Dear @TChapman500 , thank you very much for your feature request. As this is currently not on our roadmap we are very happy with PR's. So please feel free to implement the feature.

kevinawalsh, 2023-07-06:
Feature added in b22a4064abb44c6cb4a52e36cacaaeb2388e9aeb

closed 2023-07-06 by kevinawalsh