kevinawalsh/logisim-evolution #88: Feature Request. Native version for new Macs with M1 (arm)

John-Lluch, 2021-09-17:
Hi, It would be nice that a downloadable native version is provided for the new macs with the M1 (ARM) chip. I performed some research and found that the JavaSDK is available for Mac OS 11.0 with ARM 64 architecture at least from these sites: 

https://www.azul.com/downloads/?os=macos&architecture=arm-64-bit&package=jdk 
https://bell-sw.com/pages/downloads/?os=macOS&architecture=ARM

I wonder if these can be used as a starting point for implementing this

Thanks

BFH-ktt1, 2021-09-17:
dair @John-Lluch , the current V3.6.0 downloadable for MAC I build on a Mac-mini with an ARM chip, could you please specify what the problem is with this installer? This would help us debugging the issue.

kevinawalsh, 2021-09-19:
Thanks John, this is a good idea. There is also a severe mac-specific performance regression I have been hoping to track down, and it's possible it relates to the M1. So this is definitely worth trying.

I don't have an M1 available for testing, but I'd welcome any help building an testing an ARM build. I'd be happy to put the binary up alongside the existing builds. Not sure how packaging works on the M1 though.

John-Lluch, 2021-09-20:
Hi Kevin, thanks for your positive reply. I unfortunately don't have a M1 based mac either, at least not YET, but I am going to replace my 15 years old iMac soon and I am taking some time researching possible compatibility or performance issues with the apps that I use most. Logisim HC is an app that I use often, so again I really appreciate you being positive on this and eventually looking into it whenever you have the time

John-Lluch, 2023-02-17:
Hi Kevin, It's some time since I opened this Request. I hope you are well. I just want to let you know that I finally was able to invest in a new Mac Studio (with "Apple Silicon"). 

Your current version of Logisim works fine on this computer under "Roseta", and the app performs well, but it has some minor interface glitches, such as difficult to read context menu items due to lack of contrast colours in some cases, that did not happen in my old (Intel based Mac) computer.
 
The most important thing is that I am now in the disposition to perform any tests that you might ask me to do, regarding the port to ARM64, and I will be very happy to cooperate in any way that is suitable to you.
  
On a side note, I am also using other open source software that requires JAVA. Namely some KiCad Plugins. On that case, the developer instruct users to download the MacOS AArch64 version of this: https://adoptium.net/en-GB/temurin/releases/ . Not sure if this can be used as well in the case of Logisim.

Anyway, as said, I am now ready for any tests following your instructions (based on my limited knowledge) to help porting your Logisim Evolution fork, which I still regard as superior to the mainstream one,  to Apple Silicon 

Thanks for your continued support !

John-Lluch, 2023-02-17:
Ok, so I just found a logisim-evolution.jar file in the Application Bundle for version 4.0.4, and I was able to run it from the terminal using "java -jar" command, and IT WORKED ! . Running it this way also solves the interface glitches I mentioned above!, so that's great news. Launching the app this way is not as straightforward as just clicking the app icon, but it works for now. So I guess now it's just a matter of creating an actual clickable MacOS app from that. I do not know how to do that but as said, I'm happy to help in any way that I am able to. 

Thanks again and sorry for such long messages

kevinawalsh, 2023-07-04:
Under github releases, the new jar should work on Mac M1 and M2 using Java JDK 17 or above.
It must be run with one extra command-line option...
java -XstartOnFirstThread -jar logisim-evolution.jar

(And yes, the menus are on windows instead of the top of screen, and a few other mac-specific tweaks are not enabled.).

By Sept I hope to make a binary release for all platforms.
Thanks for the testing.

closed 2023-07-04 by kevinawalsh

John-Lluch, 2023-07-06:
Thanks for that. It does not appear to do anything for me with the -XstartOnFirstThread option, but it opens just fine without it and Menus are NOT in the window but on top of the screen as it normally should be on the Mac. Not sure what makes that difference for me (?), but I'm happy with it