kevinawalsh/logisim-evolution #90: Bug: Fail to launch on Linux

Codified-Vexing, 2021-12-02:
The last JAR file to run successfully on my system is version 3.1.1-HC
The current version 4.0.4hc fails to launch using OpenJDK8, OpenJDK11 or OpenJDK16
Running in command line returns the following error:
`
vex@datComputer:~/Downloads$ java '/home/vex/Downloads/logisim-evolution-4.0.4hc.jar' 
Error: Could not find or load main class .home.vex.Downloads.logisim-evolution-4.0.4hc.jar
`

Codified-Vexing, 2022-06-13:
I'm not sure that's it.

> vex@datComputer:~/Downloads$ java -jar "logisim-evolution-4.0.4hc.jar"
Error: A JNI error has occurred, please check your installation and try again
Exception in thread "main" java.lang.UnsupportedClassVersionError: JVMCFRE003 bad major version; class=com/cburch/logisim/Main, offset=6
	at java.lang.ClassLoader.defineClassImpl(Native Method)
	at java.lang.ClassLoader.defineClassInternal(ClassLoader.java:398)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:359)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:154)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:730)
	at java.net.URLClassLoader.access$400(URLClassLoader.java:96)
	at java.net.URLClassLoader$ClassFinder.run(URLClassLoader.java:1187)
	at java.security.AccessController.doPrivileged(AccessController.java:770)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:605)
	at java.lang.ClassLoader.loadClassHelper(ClassLoader.java:944)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:889)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:349)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:872)
	at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:507)

Codified-Vexing, 2022-06-17:
Yes, it seems you are right.

> vex@datComputer:~/Desktop$ java -version
> java version "1.8.0_241"
> Java(TM) SE Runtime Environment (build 8.0.6.5 - pxa6480sr6fp5-20200111_02(SR6 FP5))
> IBM J9 VM (build 2.9, JRE 1.8.0 Linux amd64-64-Bit Compressed References 20200108_436782 (JIT enabled, AOT enabled)
> OpenJ9   - 7d1059c
> OMR      - d059105
> IBM      - c8aee39)
> JCL - 20200110_01 based on Oracle jdk8u241-b07

kevinawalsh, 2022-06-17:
Glad to hear that. It's then just a matter of installing a newer Java, or
changing JAVA_HOME (or maybe updating PATH).

Codified-Vexing, 2022-06-17:
I've fixed this.
First I've typed `update-java-alternatives --list`
Then I copy one of the returned names to the following command (in my example for Java 16): 
`sudo update-java-alternatives --set "java-1.16.0-openjdk-amd64"`
It might complain about not finding some modules, but it will work nonetheless.
When I call `java -version` it shows the intended version and Logisim will launch properly.

closed 2022-06-17 by Codified-Vexing