kevinawalsh/logisim-evolution #92: Bug report: Layer depth on circuit Appearance not honoured after save

John-Lluch, 2022-04-13:
Hi Kevin, I hope you are ok.

I found a bug on the circuit appearance. I place visual elements one on top of the other by means of the commands "raise selection", "raise to top", "lower to bottom" and so on (I call this visual dept, but not sure if that's the proper term). This works fine while the project is open on Logisim. 

However after saving the project and reopening, the depth is lost and visual elements are placed in a different order. I think the bug is during project save.  

For example, consider that I have a rect and a polyline. I want the polyline to be above the rect, so I use the available commands on the appearance editor to achieve that. 

However, after saving, the project file always contains the following:
```
<polyline fill="none" points="89,80 80,80" stroke="#000000"/>
<rect fill="#f2f2f2" height="40" stroke="#000000" stroke-width="2" width="40" x="50" y="60"/>
```
Therefore when it loads the polyline is INCORRECTLY below the rect.

As a temporary workaround I am able to get it loaded correctly by editing the program file in a text editor and reordering the elements like this:
```
<rect fill="#f2f2f2" height="40" stroke="#000000" stroke-width="2" width="40" x="50" y="60"/>
<polyline fill="none" points="89,80 80,80" stroke="#000000"/> 

```
Now, the circuit loads correctly and the polyline shows correctly above the rect.

Unfortunatelly, upon saving the project again from logisim the program file is generated incorrectly again, so the bug is persistent 

I think this is because the appearance elements are stored in alphabetical order in the project xml file rather than the order (visual depth) specified by the user. Therefore, upon loading, the visual depth that the user has specifed is lost

I hope I've been able to explain this in an understandable way

Thanks

kevinawalsh, 2022-04-18:
This certainly sounds like a bug, and I can imagine where it might happen in the code. I haven't looked at the custom shape drawing code in a long time, but I will keep this open until I can.
Thanks for the report.

John-Lluch, 2022-04-19:
Thank you very much for your positive reply. I cannot emphasise enough that in my opinion your branch is the best among all from a usability standpoint: much better and faster simulation engine; realistic report of circuit oscillations, not tending to report fake or inexistent ones; useful choice of library items, particularly the invaluable file viewer; even more pleasant user interface and more

kevinawalsh, 2022-04-20:
That's really kind, thank you. It's one of my favorite projects to work on in my spare time, though regrettably I have not had any spare time to work on it for a long time now.

kevinawalsh, 2023-07-04:
John, this is now fixed. It was a careless mistake in the code on my part. I pushed a new jar file to the "release" section of github, if you'd like to try it out.
On MacOS, it must be run as: java -XstartOnFirstThread -jar logisim-evolution.jar
On Linux or Windows, it can be run as: java -jar logisim-evolution.jar

Before September I hope to put out actual binary releases for all platforms.

John-Lluch, 2023-07-06:
Hi Kevin. Thanks for that. I tried but I get the following error on the terminal : 

`joan@Joans-Mac-Studio ~ % java -XstartOnFirstThread -jar /Applications/logisim-evolution.jar
java.nio.file.NoSuchFileException: /Users/joan/.swt/swt-macos.jar
	at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)
	at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:106)
	at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)
	at java.base/sun.nio.fs.UnixFileSystemProvider.newByteChannel(UnixFileSystemProvider.java:218)
	at java.base/java.nio.file.spi.FileSystemProvider.newOutputStream(FileSystemProvider.java:484)
	at java.base/java.nio.file.Files.newOutputStream(Files.java:228)
	at java.base/java.nio.file.Files.copy(Files.java:3160)
	at com.cburch.logisim.SWTAgent.loadSWT(SWTAgent.java:98)
	at com.cburch.logisim.SWTAgent.loadPlatformSpecificSWTLibrary(SWTAgent.java:112)
	at com.cburch.logisim.SWTAgent.main(SWTAgent.java:124)
Could not load platform-specific SWT Library.
Can't extract SWT jar to /Users/joan/.swt/swt-macos.jar
java.nio.file.NoSuchFileException: /Users/joan/.swt/swt-macos.jar`

Any ideas? Thanks again

kevinawalsh, 2023-07-06:
This is a helpful report, thanks. This error is related to a new library dependency, which is supposed to get extracted automatically and stored into $HOME/.swt/swt-macos.jar (or similar file names on other platforms). It should have worked, and worked during my testing, but clearly there is something different causing it to fail for you.

I'm going to diagnose and try to replicate.

Could you try one thing. Make a folder named ".swt" by just doing:
mkdir ~/.swt

Then retry the java command. If that doesn't work, then try manually extracting the jar:
cd ~/.swt
jar xvf /Applications/logisim-evolution.jar swt-macos.jar
ls -l

This should, in principle, extract the missing swt-macos.jar from within logisim-evolution.jar, installing it into /Users/joan/.swt/

Meanwhile, I'll see if I can replicate.

kevinawalsh, 2023-07-06:
Problem replicated, diagnosed, and fixed in latest commit 119e85fc4672e82eb6afa3b7b40c48cc2b877713

I pushed a new jar to the "pre-release" section on github (july 6) that should fix the issue, I hope.

Alternatively, if you just "mkdir ~/.swt" that probably works around the issue as well.

Thanks for the bug report, this was helpful to diagnose before making a release.

closed 2023-07-06 by kevinawalsh

John-Lluch, 2023-07-09:
Hi Kevin. Thanks for your work on it. The app now launches, but unfortunately there's still some issues. 

This is what I tried: I  copied the `logisim-evolution.jar` file onto the Applications folder. Then ran the command `java -XstartOnFirstThread -jar /Applications/logisim-evolution.jar` from the terminal.

 I found the following:

- The app launches with name SWT (this appears next to the apple icon). The Logisim menu bar is actually inside the window, but the top menu items are hardly visible (all dark). When clicking on them, the menus actually drop so that indicates they are actually there, just not visible. Moving the mouse over a menu item turns it all white including the text so it can't be read.   [After a while I figured out this only happens with the dark theme selected on the Mac Preferences, with the Light Theme the menu items appear correctly and are fully readable with white text and blue background when selected. So that is workable after all by just switching the Mac theme to light before launching logisim]

- The SWT app freezes when quitting the app. A force quit is necessary for it to close. If the logistim window is closed first then SWT doesn't close either, in this case it doesn't appear as "not responding", but it must be quit anyway with 'force quit' because it won't quit with their own menu item. (I suppose this has nothing to do with logisym though) 

- The 'layer depth' bug on circuit appearance seems to still be there. The elements that I want to place one on top of another are a 'Text' and a 'State LED' (from the State tool bar item in the circuit appearance editor). I want the text to be over the LED, but after saving, closing  and opening, the text always goes behind the LED. So it appears that the bug is not solved or not completely solved.

-----

As a side comment, I think it is worth mentioning that I've been running version 4.0.4 natively in Apple Silicon for some time by doing the following: Run from terminal the command `java -jar /Applications/Logisim-HC.app/Contents/app/logisim-evolution.jar`. So basically I am calling the .jar file that is INSIDE the Mac application bundle. 

In this case, I can state the following:

- The Logisim menus appear normally on top of the Mac screen. They appear correctly regardless of the selected Mac theme 

- The application name as it appears next to the apple icon is "Logisim-evolution"  instead of "Logisim HD". Other than that there's not other visual differences that I was able to spot
 
- The app is reported by 'Activity Monitor' as "Apple" instead of "Intel"

- The app appears more responsible and simulations run MUCH faster, maybe by a factor of 3X or so. 

Thanks again

kevinawalsh, 2023-07-11:
Hi Joan,
Thanks for the details. I hadn't even realized there was a "dark mode" on
mac (I have only just transitioned to using mac more often, and until
recently I've been pretty much exclusively used linux and windows).

I've spent quite a bit of time trying to get native (or native-like) file
dialogs working on Linux, Mac and Windows, and this is the source of the
troubles you are noticing. Today I've decided after several failed attempts
to give up on the effort -- I'm going to revert back to Swing-style
dialogs. This will fix the menu issue you are seeing, and eliminate the
"-XstartOnFirstThread" requirement.

The layer ordering is a separate issue, which I'll look into more in the
coming days. If you have a small file you can share (you can send it to
***@***.*** if you prefer) and steps I should take to reproduce
what you see, that would help.

Thanks,
Kevin

kevinawalsh, 2023-07-11:
Fixed additional bugs related to this in b272e211b0725eb12f38f6ee18a35a508eefa1fd. 
There were two separate bugs I uncovered causing this issue. First was that dynamic appearance elements (like the LED) were always moved to the top-most layers when a file is opened. This was just due to how the xml reader works. The second bug was that the anchor and ports were intended to be in fixed layers, but the code for enforcing that had a bug that scrambled the order of other layers. These are now fixed.
You can find a new jar file in the release section, dated nightly-2023-07-11.

One slight change you may notice if you look closely: from now on, the anchor and ports are always the top-most elements, and can't be moved beneath other appearance elements. It would not be easy to support any other ordering due to how the simulator rendering works, so the appearance editor now just enforces this ordering at all times as well.

John-Lluch, 2023-07-12:
Hi Kekin.

I tested your changes and as far as I can tell it works correctly now !. So thank you very much again for having looked into it. About the Anchor and Ports, it's ok that they always appear on top. After that they do not appear so prominently in the circuit design editor so that's perfectly fine

I can also confirm that the app looks just right in both dark and light modes on the Mac, just the same as the official 4.0.4 release did. So that's great too.

As said, I am using this layer order feature with leds to create components (sub-circuits) that change appearance depending on state, by hiding or not perticular elements based on the led state that is on top of them. However, It would be nicer to have a feature on the circuit appearance editor that would allow users to select whether a particular visual element should be visible or not in the circuit design mode  (maybe by just changing its alpha color to transparent) depending on led state. I will write a separate feature request if that's ok.

Thanks again