kevinawalsh/logisim-evolution #93: Feature Request - Custom Dynamically changing sub-circuits appearance

John-Lluch, 2023-07-12:
It would be nicer to have a feature on the circuit appearance editor that would allow users to select whether a particular visual element should be visible or not in the circuit design mode  (maybe by just changing its alpha color to transparent) depending on led state.

Please see this as an example of what I mean 
[IdealRelay.txt](https://github.com/kevinawalsh/logisim-evolution/files/12024824/IdealRelay.txt)
This project contains a sub circuit named 1G3157 which is basically an implementation of an "ideal relay" using a multiplexer.

If you run this project from the main circuit and try to change the only input pin, you'll see what I mean. 

This is achieved by placing "leds" on top of lines, in a way that such lines get covered (hidden) or not by the leds depending on their state.

My feature request, is adding a feature in the circuit appearance editor to specify whether you want particular visual elements, such as lines, to be visible or not on the circuit design editor and during simulation, based on "state".

My "ideal relay" is just an example, but imagine the possibilities that would open to users that may want to customise their circuits appearance in more dynamic ways. I also believe that this feature would set your branch even further ahead of the other ones in terms of usability and customisability.

Thanks

John-Lluch, 2023-07-16:
Wow, just checking this and found you already committed something regarding this request. Thank you very much!. I assume you are still working on it, as I believe you did not release a .jar file for testing yet? Or maybe I am missing something... In any case no need to say I will be happy to test this feature when you are ready. So thanks again

kevinawalsh, 2023-07-19:
I just pushed a new jar file as a pre-release (look in nightly-2003-07-19 on release page).

Hope you like it! Let me know how it goes.

closed 2023-07-19 by kevinawalsh

John-Lluch, 2023-07-19:
Hi Kevin. I downloaded the referred .jar file, but I can't find the "Dynamic Visibility" tool anywhere. On the release page you wrote:
"Dynamic visibility in custom appearance (choose a shape in custom appearance, click "Dynamic Visibility" in the sidebar) lets shapes be selectively hidden based on the state of the circuit"
However I am not sure what you mean by the "sidebar" as I can't see any difference anywhere compared with the regular version. If you mean the toolbar, then it's not there(?) It must be something trivial, but I've been playing around with this release version for some time and I still can't find it. Any detailed pointers would be appreciated. Thanks

John-Lluch, 2023-07-19:
Hi Kevin, please ignore my previous comment. I was looking at the wrong place. I have tested it now and It works perfectly fine to my needs. Great addition! thank you very much!

kevinawalsh, 2023-07-21:
Implemented in 0c4973fb

John-Lluch, 2023-07-22:
Thanks for that. I found still a user interface glitch. 

To reproduce it:

- Set a Dynamic Visibility property to a shape. Click OK
- Open again the Dynamic Visibility dialog window. Select "Always Visible". Click Ok

Issue 1: After that, the Dynamic Visibility dialog can not longer be opened for the same shape

Issue 2: Even without clicking OK on the last step above, it's not longer possible to assign a Visibility property within the still opened dialog window.

Looks that after Dynamic Visibility is set back to "Always Visible", there's no way to set it again to "Depends on Circuit State"

Thanks

kevinawalsh, 2023-07-23:
Fixed in 252b3a66