kevinawalsh/logisim-evolution #96: About 5.0.0 - (minor) installation issue and something else

John-Lluch, 2023-07-21:
I downloaded the Mac .pkg file, and installed it on my Mac in the regular way. However, I found that the app is installed in the "/Applications/Logisim-HC" folder, not directly in "/Applications" top level folder. I'm not sure whether this is intended, but the problem I found is the app won't launch from inside the Logisim-HC folder. In order to work, I had to copy the "Logisim-Evolution.app" bundle to the top level /Applications folder. Once I figured it out it's not a major issue, but I thought I'd comment just in case this is not intended, or you must update something for the app to run from inside the Logisim-HD folder

Also, please check https://github.com/kevinawalsh/logisim-evolution/issues/93 . I edited my last comment a couple of days ago. Not sure if you did see it. It's my fault as I should have left an additional comment entry...

Other than that, I can not emphasize enough the excellent work and contribution that you are making to this. As a long time user of Logisim I am really pleased that you still support it as a developer, keeping binaries up to date to run on the newest hardware, while adding new useful features.

Thanks !!

kevinawalsh, 2023-07-21:
This is a strange one. No it wasn't intentional, and I'd like to fix this one as a higher priority. I can't reproduce this issue. Can you provide more info?

* I only have one mac here (running Ventura 13.2.1), and it doesn't seem to display the behavior your describe. What version are you using?

* When I download a fresh copy of the .pkg release file then install, it goes into /Applications/Logisim-Evolution.app as expected. Can you look at the directory listing below for reference, and compare it to yours?

* If I delete all installations, then install the 4.0.4 version, then install 5.0.0 on top of it, again it gets installed into the same place, simply overwriting the older version. Were you installing on top of an older version? Which version?

* Did you do anything unusual beyond clicking the installer ".pkg" file and clicking through the default options?

* I can find no mention of the string "Logisim-HC" anywhere in the project, and I don't think I would have used that string intentionally. Are you sure that's the directory name? Or was it "Logisim-Evolution-HC" ? The only thing related that comes close that I can find is the "mac-packager-identifier" which is set to "Logisim-Evolution-HC" in the jpackage parameters. But as far as I know, that option has nothing to do with the installation directory. I don't know of any way to set the installation directory, I'm just using the default jpackage tool.

Thanks for the report.

In a terminal, here is what I see right after installation if I run the command: find /Applications/Logisim-Evolution.app -type d -o -name '*.jar'

/Applications/Logisim-Evolution.app
/Applications/Logisim-Evolution.app/Contents
/Applications/Logisim-Evolution.app/Contents/_CodeSignature
/Applications/Logisim-Evolution.app/Contents/MacOS
/Applications/Logisim-Evolution.app/Contents/app
/Applications/Logisim-Evolution.app/Contents/app/logisim-evolution.jar
/Applications/Logisim-Evolution.app/Contents/Resources
/Applications/Logisim-Evolution.app/Contents/runtime
/Applications/Logisim-Evolution.app/Contents/runtime/Contents
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/_CodeSignature
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/bin
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/lib
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/lib/security
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/lib/server
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/lib/jrt-fs.jar
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/legal
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/legal/java.xml
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/legal/java.prefs
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/legal/java.logging
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/legal/java.base
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/legal/jdk.httpserver
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/legal/java.datatransfer
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/legal/java.desktop
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/conf
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/conf/security
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/conf/security/policy
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/conf/security/policy/unlimited
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/Home/conf/security/policy/limited
/Applications/Logisim-Evolution.app/Contents/runtime/Contents/MacOS

John-Lluch, 2023-07-21:
Ok, I was able to narrow down the issue. I won't affect users though, so everything is fine.

This is what happened in my case:

- I eventually still had an old Logisim-Evolution.app in the /Applications folder, but that app was not from your fork. I can't recall what fork I downloaded it as I switched to yours several years ago.
- I had your app (version 4.0.4) renamed as Logisim-HC4 in my Applications folder, but I think this is not particularly relevant
- After downloading and installing V5.0.0 .pkg file, for some reason that I do not understand, but that I can reproduce consistently every time, the .app bundle is created INSIDE a folder named "Logisim-HC". This is the app name that appears to me next to the apple icon after launch, and it's also the name that is specified as the BundleName in the app info.plist. So is this where the folder name comes from?

However, if I delete or rename the existing old, different-fork, Logisim-Evolution.app, then the installation goes correctly and a new Logisim-Evolution.app file is correctly created in the /Applications folder containing your version 5.0.0 

So this is an issue that should NOT affect other users in general unless they are installing your app while an identical named app from a different fork is already there? 

Thanks

closed 2023-07-23 by kevinawalsh