kevinawalsh/logisim-evolution #97: Serious old performance Issue - Case finally narrowed down ?

John-Lluch, 2023-07-21:
Hi Kevin,

I've been testing your "tick-test.circ" project and it gives 1.4 MHz performance on my first generation Apple Studio computer. That's fine !

The best part however is that by testing several Auto-Tick frequencies, I think I have possibly narrowed down the old issue that I described here  https://github.com/kevinawalsh/logisim-evolution/issues/74 

I hope you are able to reproduce it. This is how I consistently reproduce it.

- Open "tick-test.circ" on Logisim-HC version 5.0.0
- Select 1 KHz or 0.5 KHz as Auto-Tick Frequency
- Select Auto-Tick Enabled

The entire app will immediately turn irresponsible, there's not even need to enable the counter. The only way I found to exit that situation is to quit and open the app again

This only happens for 1 KHz, and 0.5 KHz. Lower and higher frequencies work just fine.

I hope you are able to reproduce the issue.  Something similar has affected one of my most complex projects, and assuming it's the same issue, I was never able to narrow it down or reproduce it in such a consistent way

Thank you very much.

kevinawalsh, 2023-07-21:
I can observe this behavior now.

John-Lluch, 2023-07-21:
I'm happy this can be reproduced now on a very small project. I hope this helps to narrow down the issue further. I suspect this issue is the sort of bug that must be difficult to catch... I wonder if this happens on all platforms or only Mac?

kevinawalsh, 2023-07-22:
Thanks so much for this report. I think I've found and fixed the issue. There was a poorly-conceived busy wait in one important critical section that was only getting triggered when the simulation between-tick delay was just under 1ms, which corresponded exactly to 500Hz auto-tick frequencies especially, and values close to that to a lesser extent.

I've also changed the UI update frequency from a flat 50ms (20 frames per second) to include a little jitter. This somewhat avoids the "shutter effect", where the clock and other updates appear to be frozen because the redraw timer and the simulation timer are exact multiples of each other.

I've uploaded a new jar file under the nightly tag. Any testing you can do in the next week or so would be greatly appreciated, then I'll put out a new release (pkg, zip, exe, jar) as version 5.0.1.

Thank you again.

John-Lluch, 2023-07-22:
This seem to have fixed it! 

I also tried with one of my bigger projects and I didn't find any more occurrences of the referred performance issues, so I would say everything is ok now. 

--

As a side note, after reading your comment above, I was wondering which could be the performance penalty that the actual UI refresh rate adds to the simulation.

If the UI refresh rate was in fact of any significance, my preference would be to get much lower UI refresh rate, in favour of faster overall simulation times. My point is that it doesn't really matter that the UI refreshes quickly for tick frequencies above just a few Hz because circuit states can't be fully grasped anyway in real time, and we need to step the simulation or use frequencies below 1 Hz anyway, to actually see what's really going on.

So I am referring to really slow UI refresh rates, such as once every half a second ( 2 fps), or once every 250 ms at most (4 fps). I mean, it's cool to watch the simulator visually running fast (20 fps), but if that would actually hurt project performance in a noticeable way, I'd rather sacrifice the UI in favour of real ticks per second available for the project. However, since not all users will agree with that, I suppose the UI refresh rate could be deployed as an user selectable option to avoid deviating from the current behaviour.

For example, I can think on components like the "File Viewer" configured to display assembly code (with select : Address Label) and "RAM", "ROM" components, that I use in my projects, as potentially taking significant time to update at 20 fps (??)
  
But of course, the above doesn't make sense if the current 20 fps refresh rate is not relevant in terms of time taken, which I really don't know, so in such case please disregard this comment

Thanks

John-Lluch, 2023-07-22:
Hi Kevin. There's a serious issue introduced in this particular release:  [nightly-2023-07-21](https://github.com/kevinawalsh/logisim-evolution/releases/tag/nightly-2023-07-21) 

This does not happen on Version 5.0.0, or on early versions

How to reproduce:

- Open any project, for example your tick-test
- Enable Auto-Tick
- Set Auto-Tick Frequency to something really low, for example 0.5 Hz
- Now click anywhere on the canvas. Repeat while observing the clock component

Observe that the clock component creates extra undesired ticks that match the mouse clicks. For some reason the mouse click affects directly the clock component. This can break projects because the same happens when clicking on an input button.

Thanks

kevinawalsh, 2023-07-23:
I made a fix to the simulator in 3a46128d and uploaded a new jar. I do hope that fixes this issue and doesn't introduce any new ones.
Thank you again for all the testing! I'm on vacation this week so may be a bit out of reach.

John-Lluch, 2023-07-27:
Hi, thanks for that. As far as I can tell, everything is fixed !

closed 2025-01-01 by kevinawalsh