kevinawalsh/logisim-evolution #103: Minor Bug: Should we switch to Simulate Mode while just editing a project (?)

John-Lluch, 2024-02-24:
When editing a circuit (arrow icon selected on the toolbar) while in "Design" mode on the left panel, if I right-click on a sub-circuit and then click "view [sub-circuit]", the interface switches to "Simulate" mode.

I'm not sure if this is intended, but I looks a bit unnatural to me, because when I am just editing or creating a new project from scratch, I want the left panel to stay in the "Design" mode until I explicitly run a simulation.

I believe this behaviour was not present in the very early versions of the program, but I'm not so sure.

kevinawalsh, 2025-01-01:
You are correct, early versions did not have this behavior, and there wasn't a "simulation" pane on the left either. Over the years, the simulation pane was added, and the behavior you describe was added along with it to make the simulation pane more visible and obvious. This was meant to make explicit in the interface the idea that there can be multiple simulations, with some "nested" within others. Before this time, users would poke a circuit and would often be confused upon receiving a  cryptic error about "this pin is tied to a supercircuit state" asking if they wanted to "create a new circuit state". In testing, I couldn't find any better or more clear error messages, because the entire concept of nested circuit states was invisible in the interface.

I can see how the behavior you describe is a bit unnatural, I'm just not sure how to fix it. I've tried a few things and am not happy with any of them. The main issue is that there is *always* a simulation happening. Unlike many similar tools, or things like MaxMSP, the entire program was designed to move back and forth between editing and simulation without friction.

There are FOUR ways to view a subcircuit, each a bit different: 
* Click in left design panel. Simulation changes to the most recent top-level simulation of that circuit.
* Click in left simulation panel.Simulation changes to whichever top-level or sub-simulation was chosen.
* Right-click dropdown, i.e. within some other circuit right-click and choose "view circuit". Simulation changes to a sub-simulation corresponding to the instance clicked.
* Poke tool magnifying glass, i.e. within some other circuit use the poke tool to double-click the component where the small magnifying glass appears. Simulation changes to a sub-simulation corresponding to the instance clicked.

Currently, the first two methods will stay in design mode or simulation view mode, respectively, and that seems natural. I don't think those behaviors need to change. The later two methods currently automatically switch to simulation view, to make clear you are now viewing a sub-simulation. You are describing the right-click behavior, but the poke-tool magnifying glass behavior is the same.

Some options I tried:

1. Upon right-click OR poke magnifying glass, don't switch the sidebar. This can be confusing though. You'd now be viewing a sub-simulation (where any attempt to poke a pin will be met with a cryptic error). But the left side bar would misleadingly show the design panel with the subcircuit highlighted, the same as if you were viewing a viewing a top-level simulation. This also means many people will never discover the simulation view sidebar at all.

2. When right-clicking with the arrow tool, and if the design panel is shown, then don't switch to simulation view, and don't switch to the sub-simulation. Instead, switch to the top-level simulation of that component. So arrow-tool right-click would be like clicking the left design panel, but only when the design panel is shown. This seems even more confusing.
  
3.  Change the drop-down so there are two choices, something like "view subcircuit design" and "view inner simulation". The poke tool magnifying glass would still act like "view inner simulation", as it does now.

Of all of these, I like the third one the best. Thoughts?

kevinawalsh, 2025-01-01:
I've implemented option 3, and will include that as a trial in the next release.

commit d05cb956 (2025-01-01): Popup to view design instead of nested simulation

Fixes #103

Right-click on a subcircuit now has two options: "view inner simulation"
(retains existing behavior: switch to subcircuit simulation, change left
sidebar to simulation tree); and "view subcircuit design" (new: switches
to a recent top-level simulation, does not change the left sidebar).
closed 2025-01-01 by kevinawalsh