kevinawalsh/logisim-evolution #106: Request:  Revert Click Select Behavior

TChapman500, 2025-01-27:
In v5.0.3 and prior versions, clicking on a point where multiple components overlapped would select all of the components and wires that the mouse was over.  Now this is no longer the case.  I really need this feature back as it's quite useful for quickly selecting/deselecting multiple objects at once.

kevinawalsh, 2025-01-28:
You are correct. This was changed in 1df492512a5f9bf7349659ec6a2a6c2b17699d90, so that a click in a location with multiple overlapping components will prioritize the *smallest* component, and will break ties by selecting the closest by measuring distance to center.

This was after observing students struggle to pick out small wires hidden on the edges of AND gates, and also deal with the (still present) problem of text boxes that overlap -- once overlapped, they were nearly impossible to select and separate. With the new behavior, individual components can be selected.

I suppose it would be nice to have an alternative way (perhaps option-click?) to select all components beneath the cursor. 
And perhaps a way to cycle through all of them one at a time perhaps.

TChapman500, 2025-01-28:
A cycle option and an option for the old behavior would be nice.  Perhaps have an option in preferences to set the behavior between all overlapping components and cycling through the components.